cruise-automation/webviz #655: git clone master error

daohu527, 2021-07-06:
## question
when I try to check master code, below error is appear
```
invalid path 'packages/webviz-core/migrations/frozenMigrations/2020.05.06.00:00:02.migrateNodePlaygroundNodesToObjects.js'
```

## doubt
Why can't we provide a **release version**?

1. there are 172 branches, and most are out of date, why we don’t clean it up, developers don’t need to see these branches
2. when I found a tag version, which is 2019 years later

amacneil, 2021-07-08:
Are you on windows? The colons `:` are the issue. I have [several open PRs](https://github.com/cruise-automation/webviz/pulls/amacneil) to fix cloning and building this repo on windows. [This commit](https://github.com/foxglove/webviz/commit/dfd069d0fb490e4d7882c53b6fe4e624e86832f8) fixes the migration file names, but it would need to be merged upstream.

Regarding a release version - we have a fork of webviz at [foxglove/studio](https://github.com/foxglove/studio) and have been regularly updating and publishing releases in case that is useful to you.

daohu527, 2021-07-08:
Yes, I clone on windows. Thanks for the information, it is very helpful!