cruise-automation/webviz #662: "Error: SharedArrayBuffer is not defined" while trying to play back images

KevS272, 2021-07-25:
Hello,

when I try to play back the compressed image topic of my rosbag I just get the error message 

`Error: SharedArrayBuffer is not defined`

It also says "failed to decode image on /detection_image/compressed:"
[image]

I am using Chrome version 92.0.4515.107 (Official Build) (64-bit).

I was looking on the web for the problem and it seems that google did some stuff with the SharedArrayBuffer, but I'm not familiar with the whole topic: https://developer.chrome.com/blog/enabling-shared-array-buffer/ 

Any idea how this could be fixed?

Thanks and BR
Kevin

l33tl4bs, 2021-08-05:
@KevS272 I fixed the issue making the following changes:

- packages/webviz-core/src/util/binaryObjects/binaryTranslation/index.js:178
replaced `const BufferType = self.crossOriginIsolated ? SharedArrayBuffer : ArrayBuffer;`  with `const BufferType = ArrayBuffer;`
- packages/webviz-core/src/panels/ImageView/renderImage.js:121
replaced `const notSharedData = rawData.buffer instanceof SharedArrayBuffer ? new Uint8Array(rawData) : rawData;` with `const notSharedData = rawData;`
- packages/webviz-core/.eslintrc.js:61
replaced `SharedArrayBuffer: true,` with `SharedArrayBuffer: false,`

1. unclear if all three changes are needed
2. unclear what the implications of changes are
3. maybe someone more knowledgeable can comment on this

geetesh, 2021-08-06:
> Hello,
>
> when I try to play back the compressed image topic of my rosbag I just get the error message
>
> `Error: SharedArrayBuffer is not defined`
>
> It also says "failed to decode image on [...]

Upvoting.
I am using the web-version and facing the same issue https://webviz.io/app/

sink0918, 2021-08-13:
I also have the same problem. The image can be correctly encode, but compressedImg can't.
[image]
[image]
[image]

jtbandes, 2021-08-13:
For anyone experiencing this issue, I'd encourage you to try out [Foxglove Studio](https://github.com/foxglove/studio), which renders compressed and uncompressed images without using SharedArrayBuffers.

Foxglove Studio is a visualization tool originally based on webviz, available as a cross-platform native app and also a web app. If you have any questions or issues with it, you can contact us on GitHub or [in our Slack workspace](https://foxglove.dev/slack)!

sink0918, 2021-08-14:
> For anyone experiencing this issue, I'd encourage you to try out [Foxglove Studio](https://github.com/foxglove/studio), which renders compressed and uncompressed images without using [...]

I have tried Foxglove Studio, but the plot in Studio doesn't work well as in the webviz. See in https://github.com/foxglove/studio/issues/1641.

sink0918, 2021-08-24:
When I tried the newest version of Webviz, I found this problem has been fixed.👍