cruise-automation/webviz #675: Real-time point cloud viz

Jaetriel, 2021-09-24:
Hey guys, not sure if this is the best place to post my question, but I couldn't find a better one so I'll try, would it be possible to use webviz or maybe foxglove as a starting point for a webapp which receives PointCloud2 messages from ROS generated on the host machine running a SLAM algorithm and creating a 3d map in real-time? The map is generated with 10hz frequency but I'm thinking the web visualization only needs to be updated once per second or so. I've tried using `rosbridge_websocket` already and I can get a basic example working with ros3djs, but I'm not sure if I won't be able to get better performance using worldview, and since I'd like to use React to build out the rest of the app if possible, it seems like worldview is much better integrated to work with React than ros3djs is. Right now I'm still at the point where I'm doing research and trying to figure out the best way to go about this project, so suggestions would be super helpful, thanks

jtbandes, 2021-09-24:
Hi there— I’m not sure I fully understand what you’re trying to do, but does the existing 3D panel display point clouds in the way you need? Or are you trying to build a separate, custom web app?

We’d be happy to chat more in Slack at https://foxglove.dev/slack

Jaetriel, 2021-09-27:
hi @jtbandes, the existing 3d panel looks okay but there's two things missing from it for my needs - one is an option like the "decay" time in RViz where points stick around for x amount of seconds before they stop being rendered in the scene so that you can see the whole map being built, and the second is support for Ouster lidars, since those are what we are using. My idea is to build a custom web app, yes, I'm just considering using webviz/foxglove as a starting point