cruise-automation/webviz #686: CORS issue

rfsch, 2021-11-29:
Hi

I am having issues accessing a remote bag I am hosting from a service in my k8s cluster. I am also using Keycloak gatekeeper which redirects for authentication.

I get a 404 error when webviz tries to access the remote bag:

Status code: 404.
    at exports.default.open (https://mytest.wviz.com/WorkerDataProvider.worker.js?8f2f9dfb58fc39f77292:2:538323)
    at async exports.default.open (https://mytest.wviz.com/WorkerDataProvider.worker.js?8f2f9dfb58fc39f77292:2:557592)
    at async exports.default.initialize (https://mytest.wviz.com/WorkerDataProvider.worker.js?8f2f9dfb58fc39f77292:2:478785)
    at async exports.default.initialize (https://mytest.wviz.com/WorkerDataProvider.worker.js?8f2f9dfb58fc39f77292:2:439930)

I have tried setting the CORS headers in my k8s ingress controller (nginx.ingress.kubernetes.io) as such:

nginx.ingress.kubernetes.io/configuration-snippet: |
      if ($http_origin ~ '^https?://mytest(\.(wviz|graphql|)\.com$') {
        set $allow_origin $http_origin;
      }

      if ($request_method = 'OPTIONS') {
        more_set_headers 'Access-Control-Allow-Origin: $allow_origin';
        more_set_headers 'Access-Control-Allow-Credentials: true';
        more_set_headers 'Access-Control-Allow-Methods: GET, PUT, POST, DELETE, PATCH, OPTIONS';
        more_set_headers 'Access-Control-Allow-Headers: DNT,X-CustomHeader,Keep-Alive,User-Agent,X-Requested-With,If-Modified-Since,Cache-Control,Content-Type,Authorization,X-Client-Identifier';
        more_set_headers 'Access-Control-Max-Age: 1728000';
        more_set_headers 'Content-Type: text/plain charset=UTF-8';
        more_set_headers 'Content-Length: 0';
        more_set_headers 'Access-Control-Expose-Headers: ETag, Content-Type, Accept-Ranges, Content-Length, Range';
        return 204;
      }

      more_set_headers 'Access-Control-Allow-Origin: $allow_origin';
      more_set_headers 'Access-Control-Allow-Credentials: true';
      more_set_headers 'Access-Control-Allow-Methods: GET, PUT, POST, DELETE, PATCH, OPTIONS';
      more_set_headers 'Access-Control-Allow-Headers: *';
      more_set_headers 'Access-Control-Expose-Headers: ETag, Content-Type, Accept-Ranges, Content-Length, Range';
    nginx.ingress.kubernetes.io/enable-cors: "true"

When I issue a curl -H Origin request:

curl -H "Origin: https://mytest.wviz.com" --include --verbose https://mytest.graphql.com/eed331bc5ce232158380be91d907671a14392ed9b5262c9f6d06f5bccaad1593.bag

result of curl request:

} [5 bytes data]
> GET /eed331bc5ce232158380be91d907671a14392ed9b5262c9f6d06f5bccaad1593.bag HTTP/2
> Host: mytest.graphql.com
> User-Agent: curl/7.58.0
> Accept: */*
> Origin: https://mytest.wviz.com
>
{ [5 bytes data]
* Connection state changed (MAX_CONCURRENT_STREAMS updated)!
} [5 bytes data]
< HTTP/2 307
< server: nginx/1.17.7
< date: Mon, 22 Nov 2021 14:34:29 GMT
< content-type: text/html; charset=utf-8
< content-length: 95
< cache-control: no-cache, no-store, must-revalidate, max-age=0
< location: /oauth/authorize?state=45c12482-9e6a-421e-bce5-982cac5f5b20
< set-cookie: request_uri=L2VlZDMzMWJjNWNlMjMyMTU4MzgwYmU5MWQ5MDc2NzFhMTQzOTJlZDliNTI2MmM5ZjZkMDZmNWJjY2FhZDE1OTMuYmFn; Path=/; HttpOnly; SameSite=Lax
< set-cookie: OAuth_Token_Request_State=45c12482-9e6a-421e-bce5-982cac5f5b20; Path=/; HttpOnly; SameSite=Lax
< strict-transport-security: max-age=15724800; includeSubDomains
< access-control-allow-origin: https://mytest.wviz.com
< access-control-allow-credentials: true
< access-control-allow-methods: GET, PUT, POST, DELETE, PATCH, OPTIONS
< access-control-allow-headers: *
< access-control-expose-headers: ETag, Content-Type, Accept-Ranges, Content-Length, Range
<
{ [95 bytes data]
100    95  100    95    0     0     57      0  0:00:01  0:00:01 --:--:--    57
* Connection #0 to host mytest.graphql.com left intact

I would really appreciate any hints as I just cannot find what CORS settings I am missing. Is there a way to debug to see the cause ?

Thanks!

LoganxDev, 2021-12-13:
Why do you think it's a CORS issue? Looks like it's redirecting to oath in the response. Is there a sign in process?

rfsch, 2021-12-17:
yes. when I paste the link I am providing in remote-bag-url into a browser it gets downloaded. It seems that webviz doesnt like the keycloak intercept/redirect. 

The flow is 

request file -> redirect to auth proxy keycloak -> serve file