cruise-automation/webviz #700: webviz cannot display TEXT_VIEW_FACING marker correctly

kingcansky, 2022-01-19:
I find that webviz cannot display TEXT_VIEW_FACING markers correctly even tf marker displayed in a wrong position, as picture below:

[image]

Actually, base_link tf is in the center of white box. And the position of base_link should be the id of line. I'v used Rviz to preview and Rviz displayed like this: 

[image]

Every line has its id and base_link in the right position. Then I clicked the "base_link" text, the id "2" shows and "base_link" slips to another line:

[image]

I suppose something wrong here, please check and reply.

Thx very much!

kingcansky, 2022-01-19:
BTW, if I closed text_marker, the tf_marker would display in "last tf position", which means if I remain base_link and then click "nothing" tf(I suppose a tf in (10, 0, 0)), all tf_markers will display in "base_link" position; and if remain "nothing" and then click "base_link",  all display in "nothing" position reversely. Like pic:

[image]

Confusing!

jtbandes, 2022-01-19:
Hi there – I would recommend that you try out [Foxglove Studio](https://github.com/foxglove/studio); we've recently done a lot of work to improve handling of TF frames (including TF interpolation) and you might have better luck there.

kingcansky, 2022-01-20:
> Hi there – I would recommend that you try out [Foxglove Studio](https://github.com/foxglove/studio); we've recently done a lot of work to improve handling of TF frames (including TF interpolation) and you might have better luck there. If you have further questions you can [join our Slack](https://foxglove.dev/slack).

LOL  we may try it then