cruise-automation/webviz #736: npm bootstrap and error

woxiaobaimao, 2022-03-29:
when run bootstrap, console is next：
lerna ERR! npm install --global-style --legacy-peer-deps exited 1 in 'webviz-core'
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! root@ bootstrap: `npm install && lerna bootstrap --hoist "{react,react-dom}" -- --legacy-peer-deps`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the root@ bootstrap script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/slam/.npm/_logs/2022-03-29T17_30_37_641Z-debug.log

haibin-007, 2022-07-27:
Has the problem been solved？

Niceyaa, 2022-07-29:
> 问题解决了吗？

你好  我也遇到同样的问题了  请问一下解决了吗

jtbandes, 2023-05-18:
Hi folks, have you tried [foxglove/studio](https://github.com/foxglove/studio) instead?

wuufeii, 2023-05-18:
How to run it? If I don't have Docker installed under Windows, do you have any documentation

> 
How to run it? If I don't have Docker installed under Windows, do you have any documentation

Thunder7991, 2023-07-28:
> 这个问题你们解决了吗？是node版本吗？应该使用哪个版本

我也是遇到了相似的问题, 应该是node版本的问题, 请问你解决了吗?

0815Teresa, 2024-08-29:
I have encountered the same issue. May I ask how you solved this problem?