cruise-automation/webviz #741: Could not resolve all message types

doupongzeng, 2022-06-13:
I defined some message by myself, and I can `rostopic echo /perception/camera/xxxx` in terminal, but can't show in webviz.

When I drag a rosbag from file to webviz, it's ok to show. But when I use the `http://localhost:8080/?rosbridge-websocket-url=ws://localhost:9090`, this personal topic can't show, only the other topics can show.

warning message:
```
This can happen e.g. when playing a bag from a different codebase. Message types could not be found for these topics:
/perception/camera/xxxx
/localization/xxxx
....
```

I know this issue because the special types, and use roabag-file webviz can show it, so how can I tell the webviz to recognize it when I use websocket.

tigershinny, 2022-11-23:
Some issue. Look forward to reply

CaiRugou, 2023-02-08:
Before you execute this command   "roslaunch rosbridge_server rosbridge_websocket.launch" ， should execute "source  catkin workspace/install or /devel/setup.bash".