cruise-automation/webviz #746: Error initializing player

fengpf, 2022-06-28:
I tried to drop a bag into the page but I get
Error: Expected 1 top level type definition for 'Point' but found 2
    at http://myip:8082/webvizCoreBundle.js:2:321690
    at http://myip:8082/webvizCoreBundle.js:2:321801
    at Array.forEach (<anonymous>)
    at http://myip:8082/webvizCoreBundle.js:2:321411
    at Array.forEach (<anonymous>)
    at parseMessageDefinition (http://myip:8082/webvizCoreBundle.js:2:321361)
    at ParseMessageDefinitionCache.parseMessageDefinition (http://myip:8082/webvizCoreBundle.js:448:120650)
    at exports.default (http://myip:8082/webvizCoreBundle.js:95:231010)
    at exports.default.initialize (http://myip:8082/webvizCoreBundle.js:448:165644)
    at async exports.default.initialize (http://myip:8082/webvizCoreBundle.js:448:4796)

when I drag and drop a bag into the UI

jtbandes, 2022-06-28:
Your bag file probably contains invalid message definitions. How did you create it? Have you tested to see whether it works in http://studio.foxglove.dev?

fengpf, 2022-06-29:
> yes， it works in  http://studio.foxglove.dev/