cruise-automation/webviz #747: some bags open with webviz, the bag during severe reduction

123dbl, 2022-07-05:
the bag duration in bag-database is 70 secs
[image]
but the webviz only show 35 secs data
[image]

have anybody else encountered the problem?

jtbandes, 2022-07-05:
Could you share a bag file that demonstrates this problem?

123dbl, 2022-07-11:
[apa_2022_06_28_13_55_10.bag.tar.gz](https://github.com/cruise-automation/webviz/files/9080394/apa_2022_06_28_13_55_10.bag.tar.gz)

> Could you share a bag file that demonstrates this problem? Have you tried [Foxglove Studio](https://foxglove.dev) to see if it has the same problem?

jtbandes, 2022-07-11:
```
~$ rosbag info apa_2022_06_28_13_55_10.bag
path:        .../apa_2022_06_28_13_55_10.bag
version:     2.0
duration:    41.3s
start:       Jun 27 2022 22:55:12.32 (1656395712.32)
end:         Jun 27 2022 22:55:53.60 (1656395753.60)
size:        91.9 MB
messages:    100647
```

The actual duration of the bag as reported by `rosbag info` is 41.3 sec. Webviz and Foxglove Studio both show 41 seconds of data. Seems like this is a problem with your bag database.

closed 2022-07-11 by jtbandes