cruise-automation/webviz #748: demo and demo page is broken

jxjshaw, 2022-07-09:
Hello while sharing the demo with a friend i discovered the bag wasn't loading and also the images are missing from the demo page

[image]

[image]

It looks like in both cases the files hosted on aws went missing/expired or something

cc @janpaul123 😀

jtbandes, 2022-08-04:
😄 https://studio.foxglove.dev/

jxjshaw, 2022-08-04:
lol

jxjshaw, 2022-08-04:
i dont have an aws account. can we host the bag on foxglove’s? need the
cors rule thing