cruise-automation/webviz #766: Pause button does not work

destroy314, 2023-03-29:
The pause button in the lower-left corner of the screen does not work when connecting to Rosbridge, causing memory usage to increase indefinitely and eventually crash the webpage("Aw, Snap!", error code 5). 
I'm using Chrome 111.0.5563.110 (arm64) on MacBook Pro with M1 chip.

jtbandes, 2023-03-29:
Please try [Foxglove Studio](https://github.com/foxglove/studio) instead — it is an actively maintained tool that began as a fork of this project. It has support for rosbridge as well as [foxglove_bridge](https://foxglove.dev/docs/studio/connection/using-foxglove-bridge) (a similar bridge with better performance).

Also of note is that Rosbridge is a "live" connection, and the server cannot be asked to pause incoming topics from publishers. You may need to adjust other settings in the app to ensure you aren't asking it to display an unbounded amount of data.

destroy314, 2023-03-29:
Thanks for the reply, I tried foxglove and it didn't have this problem, but I didn't find any setting to make webviz.io/app discard past data as foxglove does.