cruise-automation/webviz #769: Can this application be used as a part of custom application?

Biblbrox, 2023-05-10:
I have been developing project, which tightly works with ros2 data for communication between multiple machines. I want to integrate some visualisation capabilities into my app. I am using qt6 with qml. 

May I integrate some widgets from this application into my app without the rest? For example, for camera and lidar visualisation purposes. Does it technically possible? I mean, with qt web widget.

My application is GPLv3 licenced.

jxjshaw, 2023-05-10:
i used qt/qml like 8 years ago and i didn't recall a way to port over react components. is there some method now?

maybe if there is a way to call in a webview/iframe you could get webviz/hooked up and running and pull the viz in that way.

Biblbrox, 2023-05-17:
> i used qt/qml like 8 years ago and i didn't recall a way to port over react components. is there some method now?
> 
> maybe if there is a way to call in a webview/iframe you could get webviz/the same ros stream hooked up and running and pull the viz in that way. 

Thanks. I will try.